peterhinch/micropython-nano-gui #1: Port to interface with x,y pixel addressing....

MikeTheWatchGuy, 2018-10-01:
I really love the screenshots on your homepage!  Those are awesome looking "widgets".

I was wondering if it's possible to port the high level code so that I can use it with a tkinter canvas.  I can make the canvas essentially pixel-addressable using whatever x,y coordinate system you want.  Then perhaps that can be exposed to your graphing functions.  I can create a "draw pixel" routine easily enough.

Is your code designed in such a way it would be moved to other "Framebuffers"?

I'm looking for output widgets that can't be generally found anywhere else to can add to my GUI package, PySimpleGUI.

peterhinch, 2018-10-02:
My code uses some of the `framebuf` graphics primitives e.g. for pixels, lines, vertical lines, horizontal lines, rectangles and filled rectangles. So you'd need to implement those - an easy task as the standard algorithms are readily available (look up Brezenam's algorithm). I don't think it would be particularly hard to port, but I would be tempted to look for existing solutions. Surely someone has done something along those lines already?

It's many years since I coded for TKinter and haven't researched alternatives. As you probably know it's been around a long time. I have heard good things about PyQt.

closed 2020-06-11 by peterhinch

PySimpleGUI, 2020-06-11:
Good to know... thanks for the info.  Yea, the drawing primitives are such are indeed available in tkinter, Qt, etc.

I was asking to see if I could find a way of integrating with PySimpleGUI which would than use Qt, tkinter, WxPython and Remi (Webpage) for the final drawing.

Thanks for the follow-up.  PySimpleGUI's taken a number of twists and turns since I posted this back in Oct 2018 when it was only 3 months old. There's been nearly a million pip installs since then :-)

Thanks again Peter.