peterhinch/micropython-nano-gui #3: nanogui.py in esp8266 build error

skylin008, 2020-10-05:
Hello deterhinch! Thank you for shared the project, it is impressive! I wanted  building it forzen byte code to fimware,but any error as show follow:
GEN build-GENERIC/frozen_content.c
CC build-GENERIC/frozen_content.c
build-GENERIC/frozen_content.c:15136:1: error: unknown type name 'mp_obj_complex_t'
STATIC const mp_obj_complex_t const_obj_nanogui__lt_module_gt__conj_0 = {{&mp_type_complex}, (mp_float_t)0, (mp_float_t)1};
^
build-GENERIC/frozen_content.c:15136:1: error: braces around scalar initializer [-Werror]
build-GENERIC/frozen_content.c:15136:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__conj_0') [-Werror]
build-GENERIC/frozen_content.c:15136:1: error: initialization makes integer from pointer without a cast [-Werror]
build-GENERIC/frozen_content.c:15136:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__conj_0') [-Werror]
build-GENERIC/frozen_content.c:15136:1: error: excess elements in scalar initializer [-Werror]
build-GENERIC/frozen_content.c:15136:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__conj_0') [-Werror]
build-GENERIC/frozen_content.c:15136:1: error: excess elements in scalar initializer [-Werror]
build-GENERIC/frozen_content.c:15136:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__conj_0') [-Werror]
build-GENERIC/frozen_content.c:15300:1: error: unknown type name 'mp_obj_complex_t'
STATIC const mp_obj_complex_t const_obj_nanogui__lt_module_gt__arrow_0 = {{&mp_type_complex}, (mp_float_t)0, (mp_float_t)0};
^
build-GENERIC/frozen_content.c:15300:1: error: braces around scalar initializer [-Werror]
build-GENERIC/frozen_content.c:15300:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__arrow_0') [-Werror]
build-GENERIC/frozen_content.c:15300:1: error: initialization makes integer from pointer without a cast [-Werror]
build-GENERIC/frozen_content.c:15300:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__arrow_0') [-Werror]
build-GENERIC/frozen_content.c:15300:1: error: excess elements in scalar initializer [-Werror]
build-GENERIC/frozen_content.c:15300:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__arrow_0') [-Werror]
build-GENERIC/frozen_content.c:15300:1: error: excess elements in scalar initializer [-Werror]
build-GENERIC/frozen_content.c:15300:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__arrow_0') [-Werror]
build-GENERIC/frozen_content.c:17411:1: error: unknown type name 'mp_obj_complex_t'
STATIC const mp_obj_complex_t const_obj_nanogui__lt_module_gt__Pointer___init___0 = {{&mp_type_complex}, (mp_float_t)0, (mp_float_t)0};
^
build-GENERIC/frozen_content.c:17411:1: error: braces around scalar initializer [-Werror]
build-GENERIC/frozen_content.c:17411:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__Pointer___init___0') [-Werror]
build-GENERIC/frozen_content.c:17411:1: error: initialization makes integer from pointer without a cast [-Werror]
build-GENERIC/frozen_content.c:17411:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__Pointer___init___0') [-Werror]
build-GENERIC/frozen_content.c:17411:1: error: excess elements in scalar initializer [-Werror]
build-GENERIC/frozen_content.c:17411:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__Pointer___init___0') [-Werror]
build-GENERIC/frozen_content.c:17411:1: error: excess elements in scalar initializer [-Werror]
build-GENERIC/frozen_content.c:17411:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__Pointer___init___0') [-Werror]
build-GENERIC/frozen_content.c:17925:1: error: unknown type name 'mp_obj_complex_t'
STATIC const mp_obj_complex_t const_obj_nanogui__lt_module_gt__Dial_show_0 = {{&mp_type_complex}, (mp_float_t)0, (mp_float_t)1};
^
build-GENERIC/frozen_content.c:17925:1: error: braces around scalar initializer [-Werror]
build-GENERIC/frozen_content.c:17925:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__Dial_show_0') [-Werror]
build-GENERIC/frozen_content.c:17925:1: error: initialization makes integer from pointer without a cast [-Werror]
build-GENERIC/frozen_content.c:17925:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__Dial_show_0') [-Werror]
build-GENERIC/frozen_content.c:17925:1: error: excess elements in scalar initializer [-Werror]
build-GENERIC/frozen_content.c:17925:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__Dial_show_0') [-Werror]
build-GENERIC/frozen_content.c:17925:1: error: excess elements in scalar initializer [-Werror]
build-GENERIC/frozen_content.c:17925:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__Dial_show_0') [-Werror]
build-GENERIC/frozen_content.c:17929:1: error: unknown type name 'mp_obj_complex_t'
STATIC const mp_obj_complex_t const_obj_nanogui__lt_module_gt__Dial_show_2 = {{&mp_type_complex}, (mp_float_t)0, (mp_float_t)0};
^
build-GENERIC/frozen_content.c:17929:1: error: braces around scalar initializer [-Werror]
build-GENERIC/frozen_content.c:17929:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__Dial_show_2') [-Werror]
build-GENERIC/frozen_content.c:17929:1: error: initialization makes integer from pointer without a cast [-Werror]
build-GENERIC/frozen_content.c:17929:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__Dial_show_2') [-Werror]
build-GENERIC/frozen_content.c:17929:1: error: excess elements in scalar initializer [-Werror]
build-GENERIC/frozen_content.c:17929:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__Dial_show_2') [-Werror]
build-GENERIC/frozen_content.c:17929:1: error: excess elements in scalar initializer [-Werror]
build-GENERIC/frozen_content.c:17929:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__Dial_show_2') [-Werror]
build-GENERIC/frozen_content.c:17933:1: error: unknown type name 'mp_obj_complex_t'
STATIC const mp_obj_complex_t const_obj_nanogui__lt_module_gt__Dial_show_4 = {{&mp_type_complex}, (mp_float_t)0, (mp_float_t)0};
^
build-GENERIC/frozen_content.c:17933:1: error: braces around scalar initializer [-Werror]
build-GENERIC/frozen_content.c:17933:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__Dial_show_4') [-Werror]
build-GENERIC/frozen_content.c:17933:1: error: initialization makes integer from pointer without a cast [-Werror]
build-GENERIC/frozen_content.c:17933:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__Dial_show_4') [-Werror]
build-GENERIC/frozen_content.c:17933:1: error: excess elements in scalar initializer [-Werror]
build-GENERIC/frozen_content.c:17933:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__Dial_show_4') [-Werror]
build-GENERIC/frozen_content.c:17933:1: error: excess elements in scalar initializer [-Werror]
build-GENERIC/frozen_content.c:17933:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__Dial_show_4') [-Werror]
build-GENERIC/frozen_content.c:17934:1: error: unknown type name 'mp_obj_complex_t'
[... 11 lines cut ...]
STATIC const mp_obj_complex_t const_obj_nanogui__lt_module_gt__0 = {{&mp_type_complex}, (mp_float_t)0, (mp_float_t)3};
^
build-GENERIC/frozen_content.c:18185:1: error: braces around scalar initializer [-Werror]
build-GENERIC/frozen_content.c:18185:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__0') [-Werror]
build-GENERIC/frozen_content.c:18185:1: error: initialization makes integer from pointer without a cast [-Werror]
build-GENERIC/frozen_content.c:18185:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__0') [-Werror]
build-GENERIC/frozen_content.c:18185:1: error: excess elements in scalar initializer [-Werror]
build-GENERIC/frozen_content.c:18185:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__0') [-Werror]
build-GENERIC/frozen_content.c:18185:1: error: excess elements in scalar initializer [-Werror]
build-GENERIC/frozen_content.c:18185:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__0') [-Werror]
build-GENERIC/frozen_content.c:18186:1: error: unknown type name 'mp_obj_complex_t'
STATIC const mp_obj_complex_t const_obj_nanogui__lt_module_gt__1 = {{&mp_type_complex}, (mp_float_t)0, (mp_float_t)3};
^
build-GENERIC/frozen_content.c:18186:1: error: braces around scalar initializer [-Werror]
build-GENERIC/frozen_content.c:18186:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__1') [-Werror]
[... cut ...]
build-GENERIC/frozen_content.c:18186:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__1') [-Werror]
build-GENERIC/frozen_content.c:18186:1: error: excess elements in scalar initializer [-Werror]
build-GENERIC/frozen_content.c:18186:1: error: (near initialization for 'const_obj_nanogui__lt_module_gt__1') [-Werror]
cc1: all warnings being treated as errors
../../py/mkrules.mk:63: recipe for target 'build-GENERIC/build-GENERIC/frozen_content.o' failed
make: *** [build-GENERIC/build-GENERIC/frozen_content.o] Error 1
How to solve this issue.Thanks!

peterhinch, 2020-10-05:
Oh dear. I'm no expert on the build system but this does stand out:

> error: unknown type name 'mp_obj_complex_t'

The ESP8266 does not support the `complex` data type or the `cmath` library. I think you'll have to find out how to enable this. But there may be a good reason why this is unsupported.

As I said in the forum, it never occurred to me that nanogui could be made to work on ESP8266 so I only tested on Pyboard and ESP32, both of which support complex numbers; the module uses them fairly extensively. But I am interested in your outcome.

skylin008, 2020-10-05:
Thanks your answers!@peterhinch . I had to  look  at the esp sdk source code.Why not support complex and cmath.These lib is basic lib.

peterhinch, 2020-10-05:
I'm not entirely sure I understand your question, so I'll try to address the two interpretations I'm seeing.

As I said, I'm no expert on the build system. However I don't think the answer is in esp source. Complex math support is in the MicroPython system. Including it will require changes to files in ports/esp8266. I think there will be two things to address: ensuring that the Python VM can handle complex math and ensuring that the **cmath** library is linked in. I suggest you raise a query in the forum.

If your question is to ask why I use complex numbers it's because I find it the simplest way to do 2D rotations. By the same token I use quaternions for 3D rotation, but both can be done using trig. If I had thought that anyone would try to run **nano_gui** on such a minimal platform I would have used trig.

peterhinch, 2020-11-16:
Partly inspired by your desire to run on an ESP8266 I have completed a refactor of this library, and I think you might now be in with a chance.

The library is now a Python package so all widgets are separately loadable.  **nanogui.py** is no longer dependent on **cmath** - this dependency has been moved to the **Dial** class. So long as you don't use **Dial**, its code won't be imported and nor will **cmath**. Note that the plot demo and demos using **Dial** use complex data types.

You might be surprised at just how small **nanogui.py** is :1st_place_medal: 

if you stick to widgets other than **Dial** you should be OK.

You will need to reinstall from scratch. Application import statements must be adapted but the API and program logic is otherwise unchanged. If freezing modules (which I recommend) you will need to freeze the directory structure. Of course you only need to freeze the components you intend to use - skip unused widgets, fonts and demos to keep the size of the binary down. If you have problems here I can explain how to allocate more firmware space on the ESP. Do a **make clean** before compiling with the new package. 

I'd be very interested to hear if you have any joy.

peterhinch, 2020-11-18:
Please see [this forum post](https://forum.micropython.org/viewtopic.php?f=3&t=9090&p=52402#p52402).

tl;dr It works.

skylin008, 2020-11-20:
Thank peterhinch  very much! You are so cute!

closed 2020-12-13 by peterhinch