peterhinch/micropython-nano-gui #5: framebuf_utils.mpy runtime error

fpgirard, 2021-01-31:
Hi Peter - I really admire and appreciate all of the contributions you've been making to micropython.   I'm not sure if you want me to post an issue I'm having with nano-gui here or on the forum.  Quite willing to cross-post if that's preferable.   My issue is that I'm encountering a runtime error after cross-compiling framebuf_utils to ESP32.   Here are the steps I've taken and I'm probably doing something incorrectly.

1. Using a Lolin D32 Pro with WROVER-B, I compiled micropython with GENERIC_SPIRAM flag from source to build the latest version  - MicroPython v1.13-321-gef9fde733 on 2021-01-31; ESP32 module (spiram) with ESP32 - from scratch.   Installed this version on the ESP32 (see attached pic).  I'm fairly confident that the WROVER-B supports SPIRAM/PSRAM and I experienced zero errors compiling and deploying it on the D32.  I'm getting pretty good at doing this.
2. If I don't install a cross-compiled framebuf_utils.mpy and thus default to the STM version (which properly ignores the framebuf_utils.mpy), your aclock and other demos work flawlessly.   So does my metered 'grow.py' demo which runs indefinitely.
3. I then edit your micropython-font-to-py/writer/framebuf_utils/Makefile to use the same micropython distribution's mpy-cross and cross-compile your latest framebuf_utils to ESP32.   (ARCH = xtensawin).  No errors (see attached screen grab).
4. When I install my version in gui/core on the ESP32, the demos use the mpy executable and run for multiple iterations before crashing with the following errors:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "grow.py", line 55, in <module>
  File "grow.py", line 44, in meter
  File "gui/widgets/meter.py", line 31, in value
  File "gui/widgets/meter.py", line 56, in show
  File "gui/widgets/label.py", line 21, in __init__
  File "gui/widgets/label.py", line 32, in value
  File "gui/widgets/label.py", line 44, in show
  File "gui/core/writer.py", line 144, in printstring
  File "gui/core/writer.py", line 164, in _printline
  File "gui/core/writer.py", line 281, in _pchfast
TypeError: 
>>> 

I'm quite happy to help debug in any way that we need - realize, however, that you're dealing with an amateur and patience will be required.   Would love to get this working as I'm really looking forward to using your light gui!   Thank you.   Pascal (pascal.girard@georgetown.edu).

[image]
[image]
[image]

peterhinch, 2021-02-01:
You have had more success than me.

The reason I only distribute a `framebuf_utils.mpy` compiled for STM rather than one for each arch is that the ESP32 build does not work. In my case it crashes on import. There is [an outstanding issue](https://github.com/micropython/micropython/issues/6769) regarding native C modules on ESP32. I am waiting for this to be resolved before trying again.

In the meantime you're best off only using `framebuf_utils.mpy` on STM.

fpgirard, 2021-02-01:
Wow.  Not quite the response I was expecting but it's good to have company to join the misery.   Perhaps I should get a pyboard..... In the meantime, I'm off to implement mqtt.    Got PID flow control working nicely.   Fun...  Thanks.

peterhinch, 2021-02-02:
You should find that everything works fine without **framebuf_utils.mpy**. The only consequence is slower text rendering.

The ESP32 port is substantially slower than a Pyboard: I thoroughly recommend a Pyboard 1.1 or any Pyboard D.

If you need hardware interrupts, Pyboards are faster by a huge factor. In the case of the ESP32 with SPIRAM, interrupt latency can be well over 1000 times less on a Pyboard.

peterhinch, 2024-09-13:
Closed due to inactivity.

closed 2024-09-13 by peterhinch