peterhinch/micropython-nano-gui #6: ili9341 driver gives a blank screen

AndyThirtover, 2021-02-26:
Hi.  I'd double love to use your library.
I've tried 2 hardware versions of the ili9341 display, and get the same results on both, a blank screen.

My connections are:
CLK -> IO14
MISO -> IO12
MOSI -> IO13
LED -> IO17
RST -> IO25
DC ->IO26
CS -> IO5

Here's my color_setup.py:

from machine import Pin, SPI
import gc
from drivers.ili93xx.ili9341 import ILI9341 as SSD

pdc = Pin(26, Pin.OUT, value=0)  # Arbitrary pins
pcs = Pin(5, Pin.OUT, value=1)
prst = Pin(25, Pin.OUT, value=1)
pled = Pin(17, Pin.OUT, value=1)

gc.collect()  # Precaution before instantiating framebuf
spi = SPI(1, 10000000)
ssd = SSD(spi, dc=pdc, cs=pcs, rst=prst)

Here's what happens at REPL:
MicroPython v1.14 on 2021-02-02; ESP32 module with ESP32
Type "help()" for more information.
>>> 
>>> import gui.demos.aclock
Ignoring framebuf_utils.mpy: compiled for incorrect architecture.

I was looking to see if the SPI but gets initiated, I can see init_spi = False so I was wondering about that.   Otherwise it seems tha the code is running and the display is just white.

Hoping you can point me in the right direction.

Regards

Andy

peterhinch, 2021-02-27:
It's hard to comment other than to say that [this setup file](https://github.com/peterhinch/micropython-nano-gui/blob/master/color_setup/ili9341_setup.py) enabled the demos to work on an ESP32.

As a first step I would copy my SPI invocation line:
```python
spi = SPI(1, 10_000_000, sck=Pin(14), mosi=Pin(13), miso=Pin(12))
```

If that fails I'd try replicating my test exactly, using the file above and copying the wiring. If the test runs then you know the hardware is all OK and can then set about changing pins etc. to suit your needs.

For the record I used the reference ESP32 board without SPIRAM and an Adafruit display.

AndyThirtover, 2021-02-27:
Peter!

   Slap Like!    I just used your SPI line above and bingo!  Interesting because they should be the default pins anyway.  Issue Closed.

Double Regards

Andy

closed 2021-02-27 by AndyThirtover

peterhinch, 2021-02-27:
> they should be the default pins anyway

Indeed, this should be better documented. Glad you're up and running.