peterhinch/micropython-nano-gui #12: framebuf.GS4_HMSB format

IhorNehrutsa, 2021-04-28:
Hello, Peter.
Could you please clarify for me?
The framebuf defined as  4bit greyscale.
https://github.com/peterhinch/micropython-nano-gui/blob/090509a4d9920369a0017f4647d480acf652c969/drivers/st7789/st7789_4bit.py#L78
I guess that 
high 4bit nibble of a byte in framebuf is one pixel and 
low 4bit  nibble of a byte in framebuf is the next pixel.
Every pixel is a color number/index in the LUT.
But why they are shifted by 3 and 1 instead of 4 and 0?
https://github.com/peterhinch/micropython-nano-gui/blob/090509a4d9920369a0017f4647d480acf652c969/drivers/st7789/st7789_4bit.py#L38-L39
Thank you.

peterhinch, 2021-04-28:
Because the LUT is a 16 bit table implemented as a bytearray. Consequently successive index values increment by 2. A right shift of 3 can be seen as a right shift of 4 followed by a multiplication by 2.

IhorNehrutsa, 2021-04-28:
Yeah!
 d = (c & 0xf0) >> 3  # (c +1) * 2  # 2* LUT indices (LUT is 16 bit color) 
 e = (c & 0x0f) << 1  # c * 2 
Thanks.

closed 2021-04-28 by IhorNehrutsa