peterhinch/micropython-nano-gui #17: how could i use it to other size for epaper?

ericphilbaz, 2021-05-02:
I want to try a smaller e-ink paper(epaper2.13, just LILYGO's), but I fail，which wrong is in SSD. I change spi for epaper but nothing happened. how to fix it?

dietervansteenwegen, 2021-05-02:
@ericphilbaz I am planning to look into this as well as I have a project in mind that would use the following LilyGo display: [link](https://www.aliexpress.com/item/32869729970.html). Still, I have no idea when I will have time to do so.

It would be helpful if you would clearly define which display you would be using. Even for the one I am using, apparently there are multiple versions (yes, within version v2.3, different displays are used).

I made the following table as a reference for myself, can't remember where I got the information from:
[image]

dietervansteenwegen, 2021-05-02:
Ok, found the reference where I got that info from: (link)[https://docs.zephyrproject.org/latest/boards/shields/waveshare_epaper/doc/index.html]

peterhinch, 2021-05-03:
I have no desire to enter these shark-infested waters, not least because I have a substantial project under way.

The choice at the moment is to use an Adafruit display or to be prepared to write your own driver. As I point out in the docs, my experience with Waveshare was unsatisfactory. By contrast, in my experience, Adafruit kit works.

dietervansteenwegen, 2021-05-03:
@peterhinch that makes perfect sense. Even just the fact that a table as mine is required (undocumented different hardware options for single products) is not something that gives much hope for a smooth implementation...

dietervansteenwegen, 2021-05-03:
If I ever get around to doing it, I don't plan on asking for a lot of help, just maybe some pointers here and there. In any case, I have a (work, as in "for an employer") project that gets priority. For that one, I've chosen a adafruit eink display. That should be a good start to learn adapting and working with your library.

peterhinch, 2021-05-03:
> Even just the fact that a table as mine is required (undocumented different hardware options for single products)

Precisely why I described them as shark-infested waters. Also my experience with Waveshare. However I'll help all I can if you do write a driver.

Heliknight, 2021-05-14:
Hi - I am also looking for a little guidance - I have a 2.9" waveshare on a pi pico and was hoping that someone could assist with configuring the driver file?
https://www.waveshare.com/pico-epaper-2.9.htm

peterhinch, 2021-05-14:
It's not practical to do this without access to the physical hardware. You might get some success in [the forum](http://forum.micropython.org/index.php) where you might find someone with both the know-how and the hardware to achieve it.

Heliknight, 2021-05-14:
Thanks Peter - I will see if I can get some help there.

closed 2021-05-15 by peterhinch