peterhinch/micropython-nano-gui #19: Warning: refresh(ssd, True) called before color initialization in gui\core\colors.py

IhorNehrutsa, 2021-05-13:
Several examples have an incompletely correct initialization sequence.
They call refresh(ssd, True) but 'lut' colors is not initialized.
The more correct sequence is:
```
from color_setup import ssd  # Create a display instance
from gui.core.colors import * # Init colors
from gui.core.nanogui import refresh
refresh(ssd, True)  # clear display.
```

This is not full list of files:
micropython-nano-gui\gui\demos\aclock.py
micropython-nano-gui\gui\demos\aclock_large.py
micropython-nano-gui\gui\demos\aclock_ttgo.py
micropython-nano-gui\gui\demos\alevel.py
micropython-nano-gui\gui\demos\asnano_sync.py

It's not critical, but not clean.

peterhinch, 2021-06-09:
I take your point and I will fix this, but I'm currently fully occupied with [micro-gui](https://github.com/peterhinch/micropython-micro-gui) so anything that isn't causing an actual breakage is on the back burner.

You might be interested in [this ESP32/T-Display issue](https://github.com/micropython/micropython/issues/7374). It had me foxed but perhaps you are already aware of it.

peterhinch, 2021-06-12:
I've now pushed an update. In my view the fault was that my code required a specific import order. The patch ensures that the LUT is populated once, when `refresh` is imported. This prevents `refresh` from being called with an empty LUT.
```python
from color_setup import ssd  # Create a display instance
from gui.core.nanogui import refresh  # Color LUT is updated now.
```
Thank you for pointing out this error.

freemansoft, 2023-02-14:
@IhorNehrutsa  Can this issue be closed?

closed 2023-02-14 by peterhinch