peterhinch/micropython-nano-gui #20: Pico SPI requires "polarity=1"

goodney, 2021-05-14:
In order to get the Pico working I had to add polarity=1 as an argument to the SPI instantiation.

Also the language about the color15 demo not working on the Pico can be updated because uos.urandom() works now and the test will pass.

goodney, 2021-05-14:
I realized I underspecified this issue. Pico is connected to a 7789 240x240 board, likely from China. So perhaps the documentation could be updated to say "try adding polarity=1" if your random board from China doesn't work.

Otherwise, thanks for this library, it's great!

peterhinch, 2021-05-15:
The device datasheet (section 7.4.3) shows the clock idling at a zero level, so in my view the correct values are `polarity == 0`, `phase == 0`. In my testing the ST7789 works fine with the Pico and other platforms using these default settings.

It is, of course, possible that your board has a hardware inverter on the clock line.

closed 2021-05-15 by peterhinch