peterhinch/micropython-nano-gui #23: SH1106 support

jinzhijie, 2021-06-03:
I have a SH1106 OLED screen. Could you add the support of SH1106 drivers?

Here is the code: [robert-hh/SH1106](https://github.com/robert-hh/SH1106)

thx! ☺️

peterhinch, 2021-06-03:
I am busy on another project at the moment and will be for several weeks. Another difficulty is that it isn't practicable to develop a driver without a display, and there is a limit to how many displays I can buy :)

Converting the driver looks straightforward, the main task being to use inheritance rather than composition for the framebuffer. Perhaps you or @robert-hh might consider doing it and submitting a PR.

jinzhijie, 2021-06-03:
Your nano-gui project is very perfect and I believe that you can do it. ☺️  
Time isn't a big deal for me. I can wait for several weeks. Could you help me in your spare time?

I don't have enough time to code. I don't have the ability to submit a PR either. And robert-hh don't as familiar with this great project as you.

Besides, you can buy a SH1106 based OLED display from **Waveshare Official**. I promise that it can work properly because I am using it now.

Here is the pictures of it.

[image]
[image]

Thank you anyway!

freemansoft, 2023-02-11:
I have this display and there is a driver that looks like it might work https://github.com/raspberrypi/pico-micropython-examples/blob/master/i2c/1106oled/sh1106.py. I used it here https://joe.blog.freemansoft.com/2023/01/inland-iic-spi-13-128x64-oled-v20.html

Would you be open to a pull request that included this driver if I test with your examples?

peterhinch, 2023-02-12:
Yes.

freemansoft, 2023-02-13:
https://github.com/peterhinch/micropython-nano-gui/pull/48

freemansoft, 2023-02-14:
@jinzhijie  You should be able to close this issue. sh1106 support has been added.

closed 2023-02-14 by peterhinch