peterhinch/micropython-nano-gui #27: Pico and SPI

llvllch, 2021-11-07:
First of all, thanks a million for making such a comprehensive tool. 

I am a total novice at things this low-level, so forgive the very basic question. I'm struggling to get up and running on a pico with a (128x128) ssd1351 screen. I am thinking that my configuration of the color_setup.py may be wrong.

This is a snippet from my color_setup.py that I think may be the issue:

`
height = 128 # 1.5 inch 128*128 display

pdc = machine.Pin(20, machine.Pin.OUT)
pcs = machine.Pin(17, machine.Pin.OUT)
prst = machine.Pin(21, machine.Pin.OUT)
spi = machine.SPI(0)
gc.collect()  # Precaution before instantiating framebuf
ssd = SSD(spi, pcs, pdc, prst, height)  # Create a display instance
`

Is there a glaring error in this?

peterhinch, 2021-11-08:
It looks good but it's best to specify the initial value of pins:
```python
pdc = machine.Pin(20, machine.Pin.OUT, value=0)
pcs = machine.Pin(17, machine.Pin.OUT, value=1)
prst = machine.Pin(21, machine.Pin.OUT, value=1)
```
Are you using the right pins for SPI(0) (6 and 7)?

What actually happens when you try to run one of the demos?

llvllch, 2021-11-08:
Hey Peter, thanks for taking a look, greatly appreciated. 

The demo doesn't complain, so next step is to check whether I'm using the right pins. My two maxims for any issue with electronics: 
Check the wires
It's my fault
:)

closed 2021-11-08 by llvllch

llvllch, 2021-11-08:
Working! in color_setup.py, I replaced
`spi=machine.SPI(0)` with 
`spi = machine.SPI(0,
                  baudrate=1000000,
                  polarity=1,
                  phase=1,
                  bits=8,
                  firstbit=machine.SPI.MSB,
                  sck=machine.Pin(18),
                  mosi=machine.Pin(19),
                  miso=machine.Pin(16))`