peterhinch/micropython-nano-gui #29: upip support

petrkr, 2022-01-04:
Hello,
please consider upip support for easier upload to ESP32. Because rhsell have some issues with large and lot of files... So upip will safe lots of broken keyboards and USB cables ;)

```
/micropython-nano-gui> cp -r gui /pyboard/lib/
timed out or error in transfer to remote: b''
/micropython-nano-gui> ls gui/
```

peterhinch, 2022-01-05:
I'm puzzled: I have had no problems with `rshell` on ESP32. I don't recall anyone else experiencing this, but a forum search might yield results. I use the following args to connect to ESP32:
```bash
$ rshell -p /dev/ttyUSB0 --buffer-size=30 -a
```

I've considered putting things on PyPi but I'm not keen: I maintain a lot of repos and copying them all to PyPi would add to the maintenance burden.

petrkr, 2022-01-05:
I see.
For now I did .tar from those files and upload by using our "deploy" script. So I figured out how to do.

Also would be nice if that library will be namespaced in subdir like

nanogui.core
nanogui.driver.....

because now it's flat along other libraries and it's messy.

For more projects on pypi would be nice to use some Travis, which will do lot of job for you. For me I have only few packages now, so I do it by hand, but if we will maintain more packages, I will try to thing about travis (or some other CI) too

anyway regards for really good project

closed 2022-12-14 by petrkr

peterhinch, 2022-12-14:
Please note that upip is now obsolete. See [official docs](http://docs.micropython.org/en/latest/reference/packages.html).

petrkr, 2022-12-14:
Thanks, I somehow missed this and still creating and uploading my packages to Pypi. Compiled version from mpi seems to be better way, I have to study how to migrate those packages there.