peterhinch/micropython-nano-gui #30: Ref README.md - photographing  OLED display images

PaulskPt, 2022-01-05:
Hi Peter, I just read the contents of README.md from this repo because I want to try to use this repo.
In this README.md you wrote: ``` OLEDs are visually impressive displays with bright colors, wide
viewing angle and extreme contrast.  For some reason I find them hard to photograph. ```.

I am not a photographer. Regularly I take photographs of hardware setups and of images on displays. Usually, to be quick, I use my iPhone to take photographs. My experience using this mobile phone to take photographs of displays is not overwhelmingly positive. I also own a ```Canon PowerShot G7X Mk II``` camera. With that camera I have much better success in making photographs of displays. The Canon Camera can deal better with the big brilliance of the OLED displays becaus you can set diaphragma, shutter time etc. The camera of the mobile phone is more limited, resulting in blurry characters (as shown below) and sometimes shows artifacts too.

Below a comparison: image 1 taken with an iPhon8Plus; image 2 taken with the Canon PowerShot mentioned above.

Image 1:
[image]

Technical details of image 1:

[image]

Image 2. I did not take too much time adjusting settings of the Canon camera. I am sure that if I did, the quality of the image would be even better.

[image]

Technical details of image 2:

[image]

peterhinch, 2022-01-06:
I have recently learned a bit about photographing displays. The trick is to put everything in manual mode, shutter speed, aperture and focus. Clamp everything securely and avoid very short exposures. Here is an example.

[image]

PaulskPt, 2022-01-06:
Excellent!

peterhinch, 2022-01-08:
A couple of points I missed. Ensure a minimum amount of ambient light: I photograph in evening, but you could devise some kind of shade. Secondly, if colors are washed-out, there is local overexposure: reduce aperture.

Good luck!

closed 2022-01-08 by peterhinch