peterhinch/micropython-nano-gui #33: rsync command does not process source argument as a full rsync

jose1711, 2022-08-07:
In full `rsync` this:
```
rsync -av gui /dest  # copy what's inside gui directory
```
is not equal to,
```
rsync -av gui/ /dest  # create gui/ inside /dest, then recursively copy its contents
```
Rsync command does not really differentiate between the two (always contents is copied and not the directory itself), which is somewhat confusing.

peterhinch, 2022-08-07:
I don't understand this issue. At no point in my docs do I suggest the use of `rsync`.

jose1711, 2022-08-07:
Sorry @peterhinch, wrong repository. Anyway I know that you suggest to use `cp -r` instead but unlike `rsync` it does not give any visual feedback and first time I used it all it did was locked the terminal for approx. 10 minutes after which I decided to physically disconnect the microcontroller. Perhaps - once the issue with `rshell`'s `rsync` is fixed you can update the `README` (?). Closing.

closed 2022-08-07 by jose1711