peterhinch/micropython-nano-gui #45: Help for 4bit

vahithosan, 2023-01-03:
Hello. I have st7789 and st7735 lcd screens.
st7789 lcd 240x240 pixels
st7735 lcd 128x160 pixels (actually 132x162 ones)

Screen works when I use st7735r.py for st7735.

When I use st7735r_4bit.py for st7735 the screen does not work.

When I use st7789_4bit.py for st7789 the screen does not work.

esp32-wroom-32d
micropython version MicroPython v1.19.1 on 2022-06-18; ESP32 module with ESP32

```
from st7789_4bit import *
from machine import Pin, SPI
import gc

SSD = ST7789

pdc = Pin(2, Pin.OUT, value=0)  # Arbitrary pins
pcs = Pin(15, Pin.OUT, value=1)
prst = Pin(4, Pin.OUT, value=1)

gc.collect()  # Precaution before instantiating framebuf
spi = SPI(1, 30_000_000, sck=Pin(14), mosi=Pin(13), miso=None)
ssd = SSD(spi, height=240, width=240, dc=pdc, cs=pcs, rst=prst)

GREEN = ssd.rgb(0, 255, 0)
RED = ssd.rgb(255,0,0)
BLUE = ssd.rgb(0,0,255)
WHITE = ssd.rgb(255, 255, 255)
BLACK = ssd.rgb(0,0,0)

ssd.fill(WHITE)
ssd.text("HELLO",10,10,RED)
ssd.show()
```

peterhinch, 2023-01-04:
To get any screen working I *strongly* recommend following [these instructions](https://github.com/peterhinch/micropython-nano-gui#23-verifying-hardware-configuration). If the result is pixel perfect, nanogui will run.

vahithosan, 2023-01-04:
I did as you said. it worked very well. thanks

closed 2023-01-04 by vahithosan