peterhinch/micropython-nano-gui #47: No way to make monochrome test file working with my epaper 2.9 inch

labodezao, 2023-02-03:
Hi,
I ordered a 2.9 inch epaper from https://github.com/WeActStudio/WeActStudio.EpaperModule
And I did not manage to make it working with you code, wheras the driver seems verry similar to the waveshare code.
I test other epaper to work withy my pyboard but nothing works (nothing happends on the screen the screen) but no error in the code.
Do you have any idea ?

labodezao, 2023-02-03:
In the micropython forum, Thete where an attempt to make a driver but it is not accessible anymore 👍 https://forum.micropython.org/viewtopic.php?t=2088

labodezao, 2023-02-03:
I found the driver here, but it is possble to adapt it to nanogui : ? https://github.com/mcauser/deshipu-micropython-ili9341/blob/master/ssd1606.py

peterhinch, 2023-02-03:
It could be done: see [this reference](https://github.com/peterhinch/micropython-nano-gui/blob/master/DRIVERS.md#7-writing-device-drivers) for the approach to take. I'm not in a position to do this as I don't have one of these displays.

closed 2023-02-03 by peterhinch