peterhinch/micropython-nano-gui #49: Rotation for SSD1351

polygonfuture, 2023-03-22:
Hi Peter,  Thank you so much for your work on these libraries.    

I'm looking to rotate the SSD1351 by 180 degrees (for a physically upside down display).   
I noticed 1327 has the ability to be rotated.  Though its driver looks in some ways more simplified / abstracted compared to SSD1351.   

Do you have a suggestion of how I might achieve a 180 rotation of a 1351 OLED RGB screen?  (Its 16bit color model).

peterhinch, 2023-03-23:
There are potentially two ways to do rotation. One is by sending commands to the driver chip and the other way is by adapting the code that outputs the frame buffer to the chip. The first requires studying the datasheet to find out what is supported. The second is fairly straightforward for 180°: you need to output rows in reverse order and also reverse the order in which pixels are output to columns.

polygonfuture, 2023-03-25:
For modifying the framebuffer, would this be done within the show() function within the ssd1351_16bit.py for example?

peterhinch, 2023-03-26:
The `.show()` method would perform the rotation. The method of transferring the data to the SPI bus would be adapted to output rows and columns in the correct order.

polygonfuture, 2023-04-18:
I solve this issue sending the proper byte command to the SSD1351 chipset.  

As peter mentioned, the datasheet has a command table for a byte command to send to the SSD1351 chipset.  The chipset I have is manufactured by Solomon Systech.   

You send a Set Re-Map command listed on page 32 of the [datasheet](https://www.waveshare.com/w/upload/a/a7/SSD1351-Revision_1.5.pdf).   

Based on the table, you do this with two commands, the first tells the chipset we are going to send a Set Remap command.  With this command we send a 7 bit value that controls 6 different functions on the SSD1351. 

```
self._write(b'\xA0', 0)
self.write(b'\x76,1 1)
```
In my case I used 0x76 which translates to 01110110 command in the table.     Each binary bit represents a specific command the chipset interprets.    This correctly flips columns and rows to give a 180 degree visual rotation from the user perspective.

peterhinch, 2023-04-18:
Thanks for posting this, it is a useful resource if the requirement crops up again.

closed 2023-04-18 by peterhinch

tjlynchny, 2024-03-03:
I'm working with a Waveshare 1.27" rgb OLED screen (128 x 96) that uses the ssd1351 driver.  I, too, needed to rotate the display 180 degrees, but when I tried polygonfuture's mod for the driver, things did not quite work correctly.  I dug around in the same datasheet and came up with a fix:  _In addition to_ polygonfuture's two mod lines, I added the following which alters the Display Start Line (top of page 33 of the datasheet):

```
        self._write(b'\xA1',0)  # set command: 'Set Display Start Line'
        self._write(b'\x60',1)  # set start line to 96
```
Peter, I see you've marked this closed, but I hope you can add this fix to the thread.

And a huge thank you for nanogui and your other fantastic micropython libraries. They make working with micropython the absolute best.

-tim