peterhinch/micropython-nano-gui #50: Textbox and border

vahithosan, 2023-04-13:
Hi I can't delete borders of bordered textboxes. I tried some things but I couldn't do it

```
refresh(ssd, True)  # Initialise and clear display.
pargs0 = (3, 4, 124, 1)
pargs1 = (43, 4, 124, 1)
pargs2 = (63, 4, 124, 1)
pargs3 = (83, 4, 124, 1)
pargs4 = (103, 4, 124, 1)
pargs5 = (123, 4, 124, 1)
pargs6 = (143, 4, 124, 1)

#
tbargs0 = {'fgcolor' : WHITE,
          'bdcolor' : BLACK,
          'bgcolor' : BLACK,
         }
#
tbargs1 = {'fgcolor' : WHITE,
          'bdcolor' : RED,
          'bgcolor' : BLACK,
         }
refresh(ssd, True)
wri = CWriter(ssd, eux15, WHITE, BLACK, verbose=False)
s = '''The textbox displays'''
tb1 = Textbox(wri, *pargs1, clip=False, **tbargs1)
tb1.append(s, ntrim = 18, line = 0)
refresh(ssd)
```
Then I replace tbargs0 with tbargs1. there is no change.

```
tb1 = Textbox(wri, *pargs1, clip=False, **tbargs0)
tb1.append(s, ntrim = 18, line = 0)
refresh(ssd)
```
no change
```
tb1.bdcolor=BLACK
tb1.append(s, ntrim = 18, line = 0)
refresh(ssd)
```
no change.
```
tb1.has_border=False
tb1.append(s, ntrim = 18, line = 0)
refresh(ssd)
```
no change.

peterhinch, 2023-04-14:
The `textbox` widget is not really designed for dynamic color changes: see the `Label` widget which is. However the effects you are looking for can be achieved:
```python
tb1.bdcolor=False  # Remove border
tb1.show()  # Refresh object
refresh(ssd)
tb1.bdcolor=YELLOW
tb1.show()  # Refresh object
refresh(ssd)
```

vahithosan, 2023-04-14:
Thank you very much. 
First I worked on the Label widget. 
With Textbox, I can print what I want on a 124 pixel long area and change the back color. 
It was easier for me to work with Textbox.

closed 2023-04-14 by vahithosan