peterhinch/micropython-nano-gui #52: Pico E-Paper 4.2, Displaying Grey

CloudEater11, 2023-06-15:
I wanted to bring your attention to a bug I encountered while working with my Pico-e-Paper-4.2 gray-scale display. Specifically, correctly rendering widgets on the screen. Because the widgets use the monochrome method (bg=0, fg=1), they would display with white backgrounds (0) and light gray foregrounds (1). Again, because it used the monochrome method, overriding each label (eg. fgcolor=3), did not effect the color of text. I did some digging and unfortunately it doesn't look easy to fix, but I got a temporary flex-tape solution by editing my driver files as seen below.
### Original
```
# R21
EPD_grey_lut_ww = b".....etc
# R22H r
EPD_grey_lut_bw = b".....etc
# R23H w
EPD_grey_lut_wb = b".....etc
#  R24H b
EPD_grey_lut_bb = b".....etc
```
### Edited
```
# R21
EPD_grey_lut_ww = b".....etc
# R22H r
EPD_grey_lut_bw = b".....etc
# R23H w
EPD_grey_lut_bb = b".....etc
#  R24H b
EPD_grey_lut_wb = b".....etc
```

I switched the lut_bw with lut_bb so that it gives light grey to the value bb and black to the value wb, so that the light grey color (actually black now) continues to be passed on as 1, so that the default rendering is black-on-white. I assume there's probably a fancier solution that can be worked out, and just wanted to make you aware :). Also, it's possible I could have missed the instructions on how to display widgets in different shades of grey, but those would be a welcome addition to the e-paper section in the driver files. Great job on the code btw :)

peterhinch, 2023-06-17:
I'm not sure I follow this - apologies in advance if I am misunderstanding.

If you want a black and white display (no greyscales) then you can code for B/W and use the monochrome driver.

To show greyscales with the greyscale driver you need to code as if for a color display. You can actually run the color demos, but the mapping of colors onto greys is not ideal. Inevitably with such a small palette you need to define exactly what you want to show. You might want to define two extra colors, `LIGHTGREY` and `DARKGREY` and draw your widgets using those (plus`BLACK` and `WHITE`).

If this answers your question please let me know and I'll clarify in the docs.

peterhinch, 2024-09-13:
Closed due to inactivity.

closed 2024-09-13 by peterhinch