peterhinch/micropython-nano-gui #68: 4.2" EPD V1 discontinued, Pico now shipping with V2

ZedOud, 2024-07-02:
Hello, after I saw this PR request #64 I was concerned. As I understand, Good Display (the EPD supplier to Waveshare) is a, euphemistically speaking, "forward looking company" (meaning they discontinue product versions they see as obsolete).

If you look at (one of the many) storefronts for Good Display, they no longer list the V1 version (lacking fast refresh), and if you have a direct link to the product listing for v1 displays, some of those storefronts still had some in-stock last I checked, but the V1s are clearly discontinued. My understanding is that the Pico package from Waveshare was the only remaining way to purchase the V1 displays (my speculation: a higher quality binned version). They have hence run out of stock.

To confirm this, I just (June 25, 2024)  purchased 2x of each format of the BW 4.2" Waveshare EPD: Pico (SKU 20344), module (SKU 13353), and bare (SKU 13186). All of them arrived (June 28, 2024) with the V2 sticker on the back. 

The last time, a while ago, when I purchased the 4.2" Pico format [from this link](https://www.waveshare.com/pico-epaper-4.2.htm) that you provide on your [compatibility page](https://www.waveshare.com/pico-epaper-4.2.htm) I received V1 displays. 
Now however, I have received only V2 displays.

A few months ago, I purchased and received the V2 bare displays (and used them with the [Waveshare Pico-ePaper-Driver-Board](https://www.waveshare.com/wiki/Pico-ePaper-2.9-D) and [Adafruit ThinkInk](https://learn.adafruit.com/adafruit-rp2040-feather-thinkink/pinouts) EPD Pico/RP2040 adaptor boards) and I can report they have the same if not better partial refresh clarity and maybe better contrast (V2 has a more consistent whiter white across all units so far).

I'd be happy to answer any questions about my research into the availability of the V1 4.2" EPD displays, the quality difference between the various version of 4.2" EPDs (I have most of them: the bare & module V1 Yellow; the bare & module & Pico versions of BW V1 & BW V2; Pico V2 Red; the bare integrated touch with front-light), or any other questions about the situation you might have.

peterhinch, 2024-07-03:
Thanks for the information. I would appreciate some clarification. You bought displays from [this link](https://www.waveshare.com/pico-epaper-4.2.htm) and received V2 displays. Questions:
* Are they physically similar to the V1 displays i.e. as per the picture in the link with the white connector for connecting other targets?
* Did you test with the driver in PR#64? If not, what did you use?
* The V1 display offered such good partial refresh that I recommended it for [micro-gui](https://github.com/peterhinch/micropython-micro-gui). Have you compared ghosting levels between these displays?

I'll reopen the PR - I guess I'm going to have to buy one of these things as I can't support code without the ability to test it. I much prefer Adafruit products as they have lower levels of version churn.

ZedOud, 2024-07-04:
1. The V1 and V2 bare Displays are nearly identical to the point of being difficult to distinguish a nicer V1 from a V2 (V1 whites aren't as white). The PCB color is slightly different on the Pico and Module versions than even a recent Pico Red V2 I had ordered a few months ago. You can see on the spot the version stickers go, the V2 PCB no longer list "B/W/Y" because Yellow has been discontinued too I think (or at least Waveshare has dropped Yellow).

2. I used the Waveshare's [vanilla micropython implementation](https://github.com/waveshareteam/Pico_ePaper_Code/blob/main/python/Pico-ePaper-4.2_V2.py). I will post my version of the file. I have not tested PR #64, I'll post back here once I have.
3. The V2 partial screen refresh is very clean. In my version of Waveshare's vanilla uPy code I tested this partial screen refresh in three ways: 
- a spinning line that overlaps itself (much like a minute hand)
- an overlapping box that steadily moves across itself
- two sets of numbers incrementing to 30. 

I ran my example code on Pico Waveshare board and the Adafruit ThinkInk board with identical results.

I attempted to get a side-by-side of the equivalent code running on a V1 display, but Waveshare's vanilla uPy implementation of Partial for V1 is bad (as a comment in their own code asserts). Waveshare's [V2 nano-GUI driver implementation](https://github.com/waveshareteam/Pico_ePaper_Code/blob/main/pythonNanoGui/drivers/ePaper4in2V2.py) doesn't seem to have implemented partial screen refresh. I compared my experiments to your partial screen refresh clock and the ghosting is similarly minimal. I've actually had a Pico V1 display running your micro-GUI async demo for weeks without full-screen refreshing and the ghosting doesn't prevent it from being read, it's very nice. I'll try to set up a visually similar experiment with the V2 displays.

Also, the V2's new fast refresh mode looks just as good as a full screen refresh.

Pictures in a from a microscope of the 3 partial refresh tests.
[image]
[image]
[image]
[image]

This picture is both showing the results of the partial screen refresh experiment, as well as showing it perfectly overlapping with the V1 bare display it is sitting on.
[image]

The V1 display is interesting in how it is more see-through from behind.
[image]

Finally, here is a layout of all the versions of the 4.2" EPD I have. I highly recommend the Adafruit ThinkInk, my code includes the pin numbers and SPI settings for using it.
[image]

The code:
[Pico-ePaper-4.2_V2-partial-test.py.zip](https://github.com/user-attachments/files/16092781/Pico-ePaper-4.2_V2-partial-test.py.zip)

ZedOud, 2024-07-04:
About an hour ago I'm very sure I've gotten a V2 to display the micro-GUI epaper demo properly once (might have mistaken it for a V1?) using PR #64 using [this download of it](https://github.com/surdouski/micropython-nano-gui/blob/waveshare-4_2_rev_2_2/drivers/epaper/pico_epaper_42_v2.py) (presumably the same as the PR code?) but I haven't been able to reproduce that success since. I'll be working on this tomorrow, so I'll keep this updated.

peterhinch, 2024-07-04:
@ZedOud Thanks for that. Waveshare don't make it easy to distinguish their products. However there is an obvious silk screen difference between V1 and V2. My usual supplier only has V1 boards so I've ordered one direct from Waveshare EU. Hopefully I'll get a V2 board soon.

kingtimm, 2024-07-04:
> About an hour ago I got the V2 to display the micro-GUI epaper demo properly once with a few tweaks of hardware_setup.py and tweaks of PR #64 from [this download of it](https://github.com/surdouski/micropython-nano-gui/blob/waveshare-4_2_rev_2_2/drivers/epaper/pico_epaper_42_v2.py) (presumably the same as the PR code?) but I haven't been able to reproduce that success since. I'll be working on this tomorrow, so I'll keep this updated. You can see in this picture, below the noise, the remnants of the micro_GUI epaper demo on the V2 display.
> 
> 
> 
> [image]
> 
> 

#64 gives that staticy screen display due to it mapping PIN8 to PIN7 instead. I'm on mobile right now, but the author made a note why. It works with either changing the driver or changing on your pin header.

peterhinch, 2024-07-13:
There is now a driver for the V2 display contributed by Michael Surdouski which we have tested on micro-gui, so closing this. Please open a new issue if you require support.

closed 2024-07-13 by peterhinch