peterhinch/micropython-nano-gui #71: Label Text rendered at wrong position

Mazzelfassel, 2024-08-29:
While running some of the Demos I encountered that Labels are rendered incorrectly because their position is set incorrectly. I'm using a Waveshare Pico-ePaper-3.7 (480x280) with [this](https://github.com/waveshareteam/Pico_ePaper_Code/blob/main/pythonNanoGui/drivers/ePaper3in7.py) driver. 

After a some testing I found out, that changing this line:
https://github.com/peterhinch/micropython-nano-gui/blob/514831a9872a8815d614c1191e9f3e94e601e489/gui/widgets/label.py#L55

to 
```python 
wri.set_textpos(dev, self.row, self.col)
```

Everything works again.

peterhinch, 2024-08-30:
You are effectively setting `rcol = 0` which is forcing a left alignment. Your proposed change is the same as calling the constructor with `align=ALIGN_LEFT` - which is the default. If you pass `ALIGN_RIGHT`, text will be right-aligned. `ALIGN_CENTRE` centres the text in the label body.

If you produce an example where this does not work, I will investigate.

Mazzelfassel, 2024-08-31:
First of all I made I typo in my previous comment, the correct solution is as follows:
```python
wri.set_textpos(dev, self.row, self.col + rcol)
```
I've attached a demo project ([DisplayTest.zip](https://github.com/user-attachments/files/16825691/DisplayTest.zip)) that reproduces the described error with some examples taken from this repo and some from the waveshare one.
Additionally when the example tries to render the compas an exception is thrown so I had to disable it.

Here is the output I get when I execute the example. When I execute the example repeatedly through VS-Code (without a soft reset) the text moves more to the right with every execution, indicating that the cursor position seems to persist between executions. A soft reset returns the output back to what can be seen in the picture.
[image]

peterhinch, 2024-09-01:
The display is wrong in a number of ways. Notably the graph is being rendered partially off-screen and the word "Seismograph" is in the wrong place. These are evidence of a driver problem. The GUI will not work correctly unless all displayable elements are fully within the screen boundary.

Unfortunately the driver you are using is provided by Waveshare. While they reference my work in the code comments, I am not the author of this driver. A display driver can be tested by following the instructions [here](https://github.com/peterhinch/micropython-nano-gui/tree/master?tab=readme-ov-file#23-verifying-hardware-configuration). It is vital that any display driver is tested in this way. I suggest you try this.

If it fails (which I suspect it will) your only way forward is to take it up with Waveshare. I don't own a 3.7 inch display so I'm not in a position to debug the driver. Alternatively, use one of the supported displays.

Mazzelfassel, 2024-09-01:
Now I realized my error. I subconsciously thought that the drivers were compatible with the current version of this repo. I think the waveshare repo uses a slightly older version resulting in the weird behaviour. As I just started using displays like these and  are not confident in updateing the driver to work witht the newest version I'm just gonna use the waveshare version. So far it seems to work without a problem.

closed 2024-09-01 by Mazzelfassel