peterhinch/micropython-nano-gui #79: versions and releases

clach04, 2025-01-04:
I can't see a version number in https://github.com/peterhinch/micropython-nano-gui/blob/master/gui/core/nanogui.py and I don;t see any [tags](https://github.com/peterhinch/micropython-nano-gui/tags)/[releases](https://github.com/peterhinch/micropython-nano-gui/releases).

Do you have plans for releases/version numbers so it is clear what version is deployed?

peterhinch, 2025-01-05:
The GUI codebase is stable: changes in recent years have been restricted to the addition of device drivers.

closed 2025-01-05 by peterhinch

clach04, 2025-01-05:
Would you accept a PR to add a version 1.0.0 tuple?

peterhinch, 2025-01-06:
Yes, I have no objections.

clach04, 2025-01-08:
Opened https://github.com/peterhinch/micropython-nano-gui/pull/82