peterhinch/micropython-nano-gui #83: ssd.demo_mode attribute for EPD class

dwarde594, 2025-01-12:
Hi. I wanted to ask about `ssd.demo_mode` boolean attribute found in the EPD class in the driver files. I noticed it is used in this setup file here for a Waveshare e-Ink display: https://github.com/waveshareteam/Pico_ePaper_Code/blob/main/pythonNanoGui/color_setup.py

What is the effect of setting `ssd.demo_mode = True`? As opposed to leaving it False? Would it be considered bad practice to have demo_mode = True for non-demo applications? Can't seem to find an explanation in the documentation, might be hiding somewhere...

Thanks.

peterhinch, 2025-01-13:
It is explained [here](https://github.com/peterhinch/micropython-nano-gui/blob/master/DRIVERS.md#524-public-bound-variables). Its purpose is to enable demos written for generic displays to run on ePaper displays. There is no harm in setting it `True` in other applications, but setting it `False` allows more flexibility. You can do other things while the display refreshes.

dwarde594, 2025-01-13:
Ah perfect, thanks. That makes sense. I had some issues when I set it False using a 3.7 inch e-ink display. Sometimes it would refresh but no text would show up, probably something wrong with my code. The docs mention using the `wait_until_ready` function before refreshing so that might be the issue. No issues when demo mode is `True` though so I might just leave it like that to avoid hassle.

peterhinch, 2025-01-14:
> The docs mention using the wait_until_ready function before refreshing so that might be the issue.

Very likely.

closed 2025-01-14 by peterhinch