peterhinch/micropython-nano-gui #85: t-display-s3 1.9 - ST7789

sgbmzm, 2025-01-20:
https://lilygo.cc/products/t-display-s3?variant=42284559827125
How can I use this with ST7789?
And then use WRITER
Here are some settings that might help, but I don't know how to use
https://github.com/russhughes/s3lcd/tree/main/examples/configs/t-display-s3
Thanks a lot

sgbmzm, 2025-01-20:
This is what I tried to do, but it doesn't display anything on the screen.
```
TFT_MOSI =   11  # (SDA on schematic pdf) SPI interface output/input pin.
TFT_SCLK =   12  # This pin is used to be serial interface clock.
TFT_CS =      6  # Chip selection pin, low enable, high disable.
TFT_DC =     7  # Display data/command selection pin in 4-line serial interface.
TFT_RST =    5  # This signal will reset the device,Signal is active low.
TFT_BL =      38  # (LEDK on schematic pdf) Display backlight control pin
TFT_RD  =      9
TFT_POWER =   15

ADC_IN =     4  # Measuring battery or USB voltage, see comment below
#ADC_EN =     14  # (PWR_EN on schematic pdf) is the ADC detection enable port

BUTTON1 =    14  # right of the USB connector
BUTTON2 =     0  # left of the USB connector

from machine import Pin, SPI, ADC
import gc

from writer import Writer

from drivers.st7789.st7789_4bit import *
SSD = ST7789

pdc = Pin(TFT_DC, Pin.OUT, value=0)  # Arbitrary pins
pcs = Pin(TFT_CS, Pin.OUT, value=1)
prst = Pin(TFT_RST, Pin.OUT, value=1)
pbl = Pin(TFT_BL, Pin.OUT, value=1)
rd = Pin(TFT_RD, Pin.OUT, value=1)
power = Pin(TFT_POWER, Pin.OUT, value=1)

gc.collect()  # Precaution before instantiating framebuf
# Conservative low baudrate. Can go to 62.5MHz.
spi = SPI(1, 240_000_000, sck=Pin(TFT_SCLK), mosi=Pin(TFT_MOSI))
# Right way up landscape: defined as top left adjacent to pin 36
ssd = SSD(spi, height=320, width=170, dc=pdc, cs=pcs, rst=prst, disp_mode=PORTRAIT, display=ADAFRUIT_1_9)
# Normal portrait display: consistent with TTGO logo at top
# ssd = SSD(spi, height=240, width=135, dc=pdc, cs=pcs, rst=prst, disp_mode=PORTRAIT, display=TDISPLAY)

# Instantiate a writer for a specific font
import arial35
wri = Writer(ssd, arial35)
ssd.fill(0)
Writer.set_textpos(ssd, 0, 0) 
wri.printstring("SNOW")

```

peterhinch, 2025-01-21:
It' not possible for me to debug hardware I don't possess. The only stand-out error is your SPI baudrate of 240MHz. I would try 30MHz.

The existing t-display setup file was contributed by a user (Ihor Nehrutsa) and required a new display type [TDISPLAY](https://github.com/peterhinch/micropython-nano-gui/blob/71da31c6b471b877761421eb7d300cff2562542a/drivers/st7789/st7789_4bit.py#L34) which he researched. Your display is quite different, so I'm afraid you'll need to go through a similar process of trial and error until you can achieve a correct image from [this script](https://github.com/peterhinch/micropython-nano-gui/tree/master?tab=readme-ov-file#23-verifying-hardware-configuration).

Good luck, and if you succeed please post your setup file and I'll publish (crediting you, of course).

peterhinch, 2025-01-21:
I've looked at this some more and I don't think my driver will work.

LilyGo recommend the [Russ Hughes driver](https://github.com/russhughes/st7789s3_mpy?tab=readme-ov-file#methods) which is for displays with a parallel interface. My drivers are for SPI displays. The LilyGo pinout diagram also suggests a parallel interface. Your options are:
* Use the Russ Hughes driver (which is a C driver and incompatible with my GUIs).
* Or create a version of my driver which uses a parallel interface.

The latter is probably reasonably straightforward.

peterhinch, 2025-01-22:
From a quick look st7789py.py is for SPI displays.

The display you have has a parallel interface, and my GUIs only support SPI (and a few I2C) displays. I suggest you take up your queries with Russ Hughes as the best source of a parallel driver and support.

Closing this as I don't feel I can help here.

closed 2025-01-22 by peterhinch