peterhinch/micropython-nano-gui #86: CYD2USB weird image

ShinoNia, 2025-01-24:
Hello. I have a ESP32-2432S028Rv3 aka CYD2USB. I've tried to use the ili9341_esp32_2432S028r.py as color_setup but only got weird image.
There's static area at the top of the display; and at the bottom is an area that only show while updating. It also seems like the image is broken/duplicated and reversed.

Video:
https://youtu.be/7pZh6o6iRII

peterhinch, 2025-01-25:
If you'd like to post your setup file I'll publish it in `setup_examples`.

ShinoNia, 2025-01-25:
> If you'd like to post your setup file I'll publish it in `setup_examples`.

Hi. Thanks for replying.
The display works now but it seems like red and blue are reversed and I don't really know how to fix it.

clach04, 2025-01-26:
Hilariously I had a similar issue today, I finally made time to play with CYD2 (CYD with exact same model number.... but has 2 USB ports) and I made a mental note to follow the instructions at https://github.com/peterhinch/micropython-nano-gui/blob/71da31c6b471b877761421eb7d300cff2562542a/DRIVERS.md?plain=1#L485 which appears to be one solution.

The other solution is to update https://github.com/peterhinch/micropython-nano-gui/blob/71da31c6b471b877761421eb7d300cff2562542a/drivers/ili93xx/ili9341.py#L98

the display for the CYD2 (appears to be) physically rotated compared with CYD1

clach04, 2025-01-26:
@ShinoNia I posted PR https://github.com/peterhinch/micropython-nano-gui/pull/87 - I've not had a chance to play with this properly. I suspect it will yield the same color issue you found with the other driver.

My observation is that the driver at https://github.com/rdagger/micropython-ili9341 is fully featured and handles colors and rotations correctly. But it's a different API.

Check out https://github.com/clach04/cyd_clocks/blob/main/rainbow-test.py and https://github.com/clach04/cyd_clocks/blob/main/cyd_wrap.py NOTE the `number_of_usb_ports` variable, it's already configured for CYD2 (only needs modification for CYD1). This works for me on CYD1 and CYD2 with reasonable colors (there are some subtle differences).

I do not (yet) have a nano version of the [rainbow test ](https://github.com/clach04/cyd_clocks/blob/main/rainbow-test.py)for color checking. I would love one 😉  The other todo item on my list is fullscreen image loading (https://github.com/clach04/cyd_clocks/tree/main/images - currently only implemented for CircuitPython which I think I'm not going to not use after playing with it).

peterhinch, 2025-01-26:
Image display is described in [this doc](https://github.com/peterhinch/micropython-nano-gui/blob/master/IMAGE_DISPLAY.md).

I will investigate the color issue.

peterhinch, 2025-01-26:
I have pushed an update. Please see https://github.com/peterhinch/micropython-nano-gui/pull/87#issuecomment-2614328637 and let me know if this fixes the color issue.

clach04, 2025-01-27:
@peterhinch this is definitely resolved for me, thanks again! I opened a new discussion for bitmaps https://github.com/peterhinch/micropython-nano-gui/discussions/88

closed 2025-01-28 by peterhinch