peterhinch/micropython-nano-gui #90: doc tools question

clach04, 2025-02-02:
Great docs, thanks for taking the time to have such good docs!

I was hoping to make a trivial change to reorder DRIVERS.md to move two ILI9xxx sections next to each other. I noted that the sections are numbered.

1. Does this mean you are reluctant to reorder? For example, ILI9341 followed by ILI94xx, or even more extensive make the "Color TFT displays" section alpha-sorted?
2. what tools are you using to generate these or is it all by hand?

For number 2 - I tend to NOT number my markdown chapters, and leave as-is (either with or without a Table of contents - the github TOC is good enough most of the time). If I want a TOC I typically use https://ecotrust-canada.github.io/markdown-toc/ (I've tried a bunch out and this sucks the least and has working GitHub link generation - https://github.com/clach04/puren_tonbo is probably my largest public readme ).

For **some** projects (a small number) I use a CSS style sheet when generating HTML (and then PDF) that auto numbers the headers and then use a plugin for numbering the auto-generated table of contents (leaving the headers intact).

peterhinch, 2025-02-03:
I am aware that those sections could be adjacent. My reluctance to renumber is for fear of breaking links. I suppose I should do a grep for links.

I write markdown manually, although my editor (Pulsar) can render the result.

clach04, 2025-02-04:
> I am aware that those sections could be adjacent. My reluctance to renumber is for fear of breaking links. I suppose I should do a grep for links.

If it's only the consistency of links in this project documentation, I think we're in good shape:

```
C:\code\cyd\micropython-nano-gui>rg -i \(.*DRIVERS.md#3.*\)   ADAFRUIT.md  ASYNC.md  DISPLAYS.md  DRIVERS.md  FPLOT.md  IMAGE_DISPLAY.md  README.md
DRIVERS.md
40: 3. [Color TFT displays](./DRIVERS.md#3-color-tft-displays)
41:  3.1 [Drivers for ST7735R](./DRIVERS.md#31-drivers-for-st7735r) Small TFTs
42:  3.2 [Drivers for ILI9341](./DRIVERS.md#32-drivers-for-ili9341) Large TFTs
43:  3.3 [Drivers for ST7789](./DRIVERS.md#33-drivers-for-st7789) Small high density TFTs
44:  &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;3.3.1 [TTGO T Display](./DRIVERS.md#331-ttgo-t-display) Low cost ESP32 with integrated display
45:  &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;3.3.2 [Waveshare Pico Res Touch](./DRIVERS.md#332-waveshare-pico-res-touch)
46:  &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;3.3.3 [Waveshare Pico LCD 2](./DRIVERS.md#333-waveshare-pico-lcd-2)
47:  &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;3.3.4 [Troubleshooting](./DRIVERS.md#334-troubleshooting)
48:  3.4 [Driver for ILI94xx](./DRIVERS.md#34-driver-for-ili94xx) Generic ILI94xx and HX8357D driver for large displays.
49:  3.5 [Driver for gc9a01](./DRIVERS.md#35-driver-for-gc9a01) Round 240x240 displays.
489:[ILI9486 driver](./DRIVERS.md#34-driver-for-ili94xx) with the `mirror`

```

Only DRIVERS.md has links to the driver chapter 3, "Color TFT displays".

If you are concerned with external documents/pages breaking links that would be a reason to not make changes.

peterhinch, 2025-02-04:
This doc applies to the other GUI's (divers are common to all). There may be links in code comments (e.g. setup examples for all GUI's).

clach04, 2025-02-05:
> This doc applies to the other GUI's (divers are common to all). There may be links in code comments (e.g. setup examples for all GUI's).

for the three projects I can locate, the only links are in drivers.md (including source code):

```
C:\code\cyd\mp_guis>dir
 Volume in drive C is Windows
 Volume Serial Number is 02C1-2C6F

 Directory of C:\code\cyd\mp_guis

02/04/2025  06:35 PM    <DIR>          .
02/04/2025  06:35 PM    <DIR>          ..
02/04/2025  06:35 PM    <DIR>          micropython-micro-gui
02/04/2025  06:35 PM    <DIR>          micropython-nano-gui
02/04/2025  06:35 PM    <DIR>          micropython-touch
               0 File(s)              0 bytes
               5 Dir(s)  590,635,614,208 bytes free

C:\code\cyd\mp_guis>rg -i \(.*DRIVERS.md#3.*\)
micropython-nano-gui\DRIVERS.md
40: 3. [Color TFT displays](./DRIVERS.md#3-color-tft-displays)
41:  3.1 [Drivers for ST7735R](./DRIVERS.md#31-drivers-for-st7735r) Small TFTs
42:  3.2 [Drivers for ILI9341](./DRIVERS.md#32-drivers-for-ili9341) Large TFTs
43:  3.3 [Drivers for ST7789](./DRIVERS.md#33-drivers-for-st7789) Small high density TFTs
44:  &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;3.3.1 [TTGO T Display](./DRIVERS.md#331-ttgo-t-display) Low cost ESP32 with integrated display
45:  &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;3.3.2 [Waveshare Pico Res Touch](./DRIVERS.md#332-waveshare-pico-res-touch)
46:  &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;3.3.3 [Waveshare Pico LCD 2](./DRIVERS.md#333-waveshare-pico-lcd-2)
47:  &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;3.3.4 [Troubleshooting](./DRIVERS.md#334-troubleshooting)
48:  3.4 [Driver for ILI94xx](./DRIVERS.md#34-driver-for-ili94xx) Generic ILI94xx and HX8357D driver for large displays.
49:  3.5 [Driver for gc9a01](./DRIVERS.md#35-driver-for-gc9a01) Round 240x240 displays.
489:[ILI9486 driver](./DRIVERS.md#34-driver-for-ili94xx) with the `mirror`
```

Where the checkouts are from:
  * https://github.com/peterhinch/micropython-micro-gui
  * https://github.com/peterhinch/micropython-nano-gui
  * https://github.com/peterhinch/micropython-touch

Any others that need checking?

peterhinch, 2025-02-05:
Thanks for that. I have pushed an update.

closed 2025-02-05 by peterhinch