peterhinch/micropython-nano-gui #91: Labels and borders

clach04, 2025-02-02:
Labels work as documented but the behavior is a little confusing.

From my reading of https://github.com/peterhinch/micropython-nano-gui/blob/77f58af1fab27e0ec6ba959a2b04cd4061fa828f/README.md?plain=1#L585

borders are shown if:

1. None is passed in
2. a color is passed in

From a quick skim, and more importantly the code for Label I saw False as a default so incorrectly assumed True would be the same as None.

https://github.com/peterhinch/micropython-nano-gui/blob/77f58af1fab27e0ec6ba959a2b04cd4061fa828f/gui/core/nanogui.py#L89..L94

Is is reasonable to also support True? I'm happy to knock something up.

peterhinch, 2025-02-03:
We have the following allowed args:  
* `False` No border.  
* `None` Use the color from the `Writer` instance, as per foreground (FG) and background (BG) colors.  
* `A color` Use that color (as per FG and BG).  

This is consistent with fFG and BG, with the exception of `False`. (It is not possible to have no FG or BG). What do you propose as the outcome if `True` is passed?

clach04, 2025-02-04:
> ... What do you propose as the outcome if `True` is passed?

Treat the same as the None case, i.e. https://github.com/peterhinch/micropython-nano-gui/blob/77f58af1fab27e0ec6ba959a2b04cd4061fa828f/gui/core/nanogui.py#L89..L90

        if bdcolor is None or bdcolor is True:
            bdcolor = fgcolor

That way it's the inverse of False. It has a border that matches the fore ground color/colour.

peterhinch, 2025-02-05:
The way colors are specified is supposed to be consistent across the three GUI's. If a color is specified, it is used. If `None` is passed a documented default behaviour occurs (in many cases getting a color from the `Writer`).

The `Label` class is something of an anomaly because border color has two common use cases that don't correspond to a specific color. `None` behaves conventionally in using the foreground color. This left the problem of how to specify no border. I decided on `False`. Arguably `None` would be more logical, but it doesn't match the behaviour of other colors and I wanted the default label to have a border whose color matched its contents.

For better or worse, `False` is the solution I adopted. Adding `True` as an alternative to `None` might confuse those accustomed to the convention that `None` produces default behaviour.

I see no reason to modify this.