peterhinch/micropython-nano-gui #95: LIGHTRED mislabeled DARKRED

clach04, 2025-02-22:
This is a very limited extract from https://github.com/peterhinch/micropython-nano-gui/pull/9#issuecomment-827606000

LIGHTRED in:

https://github.com/peterhinch/micropython-nano-gui/blob/701149bc3ad2eecff2897044225df12bcc3f07a2/gui/core/colors.py#L17
https://github.com/peterhinch/micropython-nano-gui/blob/701149bc3ad2eecff2897044225df12bcc3f07a2/gui/core/colors.py#L32

Are mislabeled, the color defined is actually a dark red, darker than the red that is declared in https://github.com/peterhinch/micropython-nano-gui/blob/701149bc3ad2eecff2897044225df12bcc3f07a2/gui/core/colors.py#L16

Based onthe  discussion in https://github.com/peterhinch/micropython-nano-gui/pull/9 sounds like there is concern about renaming.

What about the idea of adding DARKRED as an alias?

I..e.:

    ...
    DARKRED = LIGHTRED = create_color(3, 140, 0, 0)
    ...
    DARKRED = LIGHTRED = SSD.rgb(140, 0, 0)

So the color stays the same, but there would be a more accurate name.

peterhinch, 2025-02-22:
If I were starting from scratch I would name many of these colors differently. Alas these names are used across multiple GUI's and even more demos and applications so I don't plan on making changes.

Users are encouraged to create their own aliases in their applications.

closed 2025-02-22 by peterhinch

clach04, 2025-02-22:
That makes sense @peterhinch , I'd already followed your sugestion 😄 I have some PIL/Pillow  palettes set up with slightly tweaked names at https://github.com/clach04/cyd_clocks/issues/40#issuecomment-2676022881

I have a proof of concept server that generates bitmaps which the micropython client simply pulls down and displays.